alangibson/homeassistant-sungrow #1: 2 sungrow inverter

urueedi, 2022-09-19:
Thank you for your work, at the moment is not possible to add more then one sungrow inverter. I have 2 sungrow inverter 1x sg40cx + 1x sg50cx.

AlexanderLanin, 2023-12-21:
It's not working for other reasons, but I can add two:
[image]

AlexanderLanin, 2023-12-22:
Fixed the other problem (unsupported inverter, see https://github.com/bohdan-s/SunGather/issues/157).
Anyway, now I see the following:
```
Platform sungrow does not generate unique IDs. ID sungrow_16690_meter_power already exists - ignoring sensor.sh80rt_20_16690_current_meter_power
Platform sungrow does not generate unique IDs. ID sungrow_16690_total_active_power already exists - ignoring sensor.sh80rt_20_16690_current_active_power
Platform sungrow does not generate unique IDs. ID sungrow_16690_export_to_grid already exists - ignoring sensor.sh80rt_20_16690_current_power_exported_to_grid
[...]
```

AlexanderLanin, 2023-12-22:
Note: I've checked the content of "serial_number" just after scraping:
```
2023-12-22 03:43:07.753 WARNING (SyncWorker_4) [custom_components.sungrow.SunGather.inverter] (192.168.13.80:502 (2)) serial_number: 16690
2023-12-22 03:43:07.854 WARNING (SyncWorker_5) [custom_components.sungrow.SunGather.inverter] (192.168.13.79:502 (1)) serial_number: 16690
```

I don't know if that's normal...

Inverters connected via modbus.

- master: 192.168.13.79:502 (slave 1)
- slave: 192.168.13.80:502 (slave 2)

alangibson, 2023-12-22:
@AlexanderLanin I'm looking at the unsupported inverter problem now. What fix did you do?

alangibson, 2023-12-22:
Two devices getting the same serial number is very odd. In config_flow.py:

```
async def async_step_user(self, user_input=None):
    ...
    unique_device_id = inverter.latest_scrape.get('serial_number')
```

So this gets fixed at setup time. I need to look deeper on this one.

AlexanderLanin, 2023-12-22:
> So this gets fixed at setup time. I need to look deeper on this one.

I logged it directly in load_registers():
```
                    if register_name == "serial_number":
                        logger.warning(f"serial_number: {register_value}")
```

It's the same value.

AlexanderLanin, 2023-12-22:
> @AlexanderLanin I'm looking at the unsupported inverter problem now. What fix did you do?

https://github.com/bohdan-s/SunGather/issues/157#issuecomment-1867811522

AlexanderLanin, 2023-12-22:
@alangibson Ah found it. It's 10 registers, not one. And I actually get exactly the number that is printed on the side of the inverter!

```
result = client.read_input_registers(4989, count=10, unit=unit)
serial_number = "".join([chr(c >> 8) + chr(c & 0xFF) for c in result.registers])
```

PR #10 (2024-01-02): Read the device serial number correctly
As discovered in #1 the serial number is 10 registers and not a single one.
Here is one idea how to implement it.

While this doesn't address all the problems with multiple inverters, it should be enough to close #1:
[image]

As I don't have the slightest idea how to implement tests here, this code is "works for me".

Note: this will change entity ids and therefore all previously collected data is lost.

This is what it looks like after this PR:
[image]
closed 2024-01-02 by alangibson