alangibson/homeassistant-sungrow #2: {"message":"Invalid handler specified"}

kalaws, 2023-12-21:
Appreciate the work. I can’t load config flow however. Getting the above error message. 

Cheers

kalaws, 2023-12-21:
I see you just recently changed some directory name. According to below post, the error might be related to that. 

https://community.home-assistant.io/t/error-adding-integration/485237/16

kalaws, 2023-12-21:
More specific log entry:

Error occurred loading flow for integration sungrow: cannot import name 'Defaults' from 'pymodbus.constants' (/usr/local/lib/python3.11/site-packages/pymodbus/constants.py)

alangibson, 2023-12-21:
> I see you just recently changed some directory name.

Domain and directory name are both "sungrow" so it shouldn't be that.

> Error occurred loading flow for integration sungrow: cannot import name 'Defaults' from 'pymodbus.constants'

That would indicate that it's a version problem with pymodbus. Do you have the addon from https://github.com/bohdan-s/hassio-repository installed still?

AlexanderLanin, 2023-12-21:
Oh I guess that's my fault. Kind of ;-)

> Logger: custom_components.sungrow.SunGather.inverter
> Source: custom_components/sungrow/SunGather/inverter.py:111
> Integration: Sungrow ([documentation](https://github.com/alangibson/homeassistant-sungrow))
> First occurred: 12:06:30 AM (2 occurrences)
> Last logged: 12:06:30 AM

> Unknown Type Code Detected: 3602

It's an SH8.0RT-20, see https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L1970

I'm a little lost where I should create a separate issue for that?

kalaws, 2023-12-22:
No, don't have it installed. Very strange if it works for others.

kalaws, 2023-12-23:
I installed a fork of bohdan’s Hassio addon, which tried to solve the issue by allowing to “break system” related to some python stuff. Perhaps that might have something to do with it? Any idea how I might unbreak what it broke, or is it straight to restoring from backup?

kalaws, 2023-12-23:
```
# Install Python packages on system level --> PEP668
ENV PIP_BREAK_SYSTEM_PACKAGES=1

```

https://github.com/anthony-spruyt/hassio-repository/blob/5f97d9257b376ff7c1e908ef3a408f35736dcac8/SunGather/Dockerfile

alangibson, 2023-12-23:
Are you running the Home Assistant OS, or did you install Home Assistant yourself on some existing OS? If it's the latter, run `cat /usr/local/lib/python3.11/site-packages/pymodbus/version.py` to see if  you have 2.5.3 installed. Just look for this line: `version = Version('pymodbus', 2, 5, 3)`. You may be able to fix it manually with `pip uninstall pymodbus` by just deleting the pymodbus module at /usr/local/lib/python3.11/site-packages/pymodbus. 

If you're running Home Assistant OS, I've never accessed the core filesystem. They have a pretty complicated storage setup. I'd have to look into that.

kalaws, 2023-12-23:
Thanks. Unfortunately it’s the former!

kalaws, 2023-12-23:
I have no `pymodbus`folder in the `site-packages` folder. But perhaps that other addon fork installed it at some other location, not tied to usr?

kalaws, 2023-12-23:
If I try to `pip install pymodbus` that works, but the same error persists. All my py stuff is in `/usr/lib/python3.11/...` rather than `/usr/local/lib/python3.11/...`. Perhaps that's a HA OS thing? Could the search path be altered?

kalaws, 2023-12-25:
Is there any easy way to turn the integration into something that might be run in AppDaemon instead? In AD I can load all the needed requirements.

alangibson, 2023-12-28:
I'm not able to reproduce this at all, and I'm out of ideas as to how to fix it. Try posting a question on https://community.home-assistant.io/ and see if anyone there has any ideas. Maybe they'll at least know how to pull some useful debugging info for this problem.

alangibson, 2024-01-07:
If you're still having trouble, maybe give @AlexanderLanin 's fork a try. That will probably become the official version anyway.

https://github.com/AlexanderLanin/homeassistant-sungrow

closed 2024-01-07 by alangibson