alangibson/homeassistant-sungrow #14: Need-Help: development environment

AlexanderLanin, 2023-12-26:
I need some help with the development environment.
In the end we could document how to do this, so this ticket has real benefit :-)

* Option: Studio Code Server Addon
  Pro: Super Easy
  Con: no git integration (and completely different structure), not all VS-Code extensions available

* Option: Terminal & SSH Addon + Local VS Code instance connected via SSH
  Pro: all extensions available (e.g. GitHub Copilot)
  Con: python not available (no linter, no auto formatting), different structure in git

* Option: work on branch and reload integration in home assistant via HACS
  (not yet attempted)

* Option: work locally and copy files via bash script over SSH
  (not yet attempted. Sounds promising!)

alangibson, 2023-12-28:
I tend to use "work on branch and reload integration in home assistant via HACS" even though it is slow. It's only practical when you know the HA api well enough that you can be reasonably sure what you did will work when you load it up in HA.

Finding the optimal way to do it is way down on my mental TODO list. I just haven't had time for it.

AlexanderLanin, 2024-01-01:
I've followed up on option 4. It's not perfect, but actually better than the others.
Here is what I'm running as `update_ha.sh`:

```
#!/bin/bash
set -euxo pipefail

# ToDo: skip __pycache__ folders
# See e.g. https://stackoverflow.com/questions/15121337/recursively-use-scp-but-excluding-some-folders

rm -rf custom_components/sungrow/__pycache__
rm -rf custom_components/sungrow/core/__pycache__

scp -r custom_components/sungrow/ root@homeassistant.local:/root/homeassistant/custom_components || {
  echo "Error: Failed to copy files."
  echo "* Do you have ssh addon running in home assistant?"
  echo "* Did you configure it for remote access?"
  echo "* Did you add your ssh key to the authorized_keys? (run: eval `ssh-agent` and ssh- ssh-add)"
  echo "* Can you connect via ssh? (run: ssh root@homeassistant.local)"
  exit 1
}

ssh root@homeassistant.local "ha core restart"
```

I'm using the `Terminal & SSH` Add-On in HA. But I cannot find the instruction of what I had to do to configure remote access. Basically, what you need is a key pair as documented very nicely in https://docs.github.com/en/authentication/connecting-to-github-with-ssh/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent
Then you configure `Terminal & SSH` -> `authorized_keys` with the content of `id_ed25519.pub`