alangibson/homeassistant-sungrow #15: Added a HACS integration and...

FHoevi, 2023-12-27:
... now what? Sorry if this is an absolute newbie question but what should happen now? What else do I need to configure after adding the integration as instructed to be able to read values from my inverter? What am I supposed to do?

AlexanderLanin, 2023-12-27:
I'm not sure this is already usable in any kind of environment beyond development / proof of concept.

Anyway if you configured it correctly you should see entities under http://homeassistant.local:8123/config/entities?domain=sungrow
No further yaml needed.

alangibson, 2023-12-28:
I've been using this integration myself for over a year, so it should work for you as long as you just have 1 inverter. The bug with 2 inverters should be fixed soon.

I'd recommend reading the HomeAssistant docs once to get familiar with how it works. After installing the integration, you need to add a Sungrow device. Check out this video for a general walkthrough: https://www.youtube.com/watch?v=-y5ofKHDXmw

FHoevi, 2024-01-01:
@alangibson @AlexanderLanin First of all, happy new year! Thanks a lot to both of you, now measured values from my inverter are finally showing up! Alan, you were right, the only thing which was missing was to add a Sungrow device. So I thin I'm all set now, at least in respect of integrating the inverter to my Home Assistant setup. Thanks a lot guys, I will close this thread now.

closed 2024-01-01 by FHoevi