alangibson/homeassistant-sungrow #16: Sungrow SH6.0RT: NOT all parameters read properly

pawelsn, 2023-12-28:
Please let me know how can I try to fix manualy at least " total_battery_charge_energy_from_pv" - so I could have proper data in HA energy card. Everything else is read OK,

Logger: custom_components.sungrow.sensor
Source: custom_components/sungrow/sensor.py:210
Integration: Sungrow (documentation)

Sensor lookup value is not available in data array: daily_battery_charge_energy_from_pv
Sensor lookup value is not available in data array: total_battery_charge_energy_from_pv
Sensor lookup value is not available in data array: daily_export_power_from_pv
Sensor lookup value is not available in data array: total_export_power_from_pv
Sensor lookup value is not available in data array: meter_power

pawelsn, 2023-12-28:
I'm connected via http mode

pawelsn, 2023-12-31:
I will try to fix it by the change in in const.py:

#SUNGROW_DAILY_BATTERY_CHARGE_PV_ENERGY = "daily_battery_charge_energy_from_pv"
#SUNGROW_TOTAL_BATTERY_CHARGE_PV_ENERGY = "total_battery_charge_energy_from_pv"
SUNGROW_DAILY_BATTERY_CHARGE_PV_ENERGY = "daily_battery_charge_from_pv"
SUNGROW_TOTAL_BATTERY_CHARGE_PV_ENERGY = "total_battery_charge_from_pv"

based on information (output format data) traced here:
https://community.openhab.org/t/sungrow-binding/147442/4

FHoevi, 2024-01-06:
I successfully did something similar today.

But a thing which not really works is to add a new figure like `SUNGROW_BATTERY_LEVEL = "battery_level"`. How could I achieve that? It does not really show up.

FHoevi, 2024-01-06:
OK, it seems to be a combination of corresponding entries in two distinct files: `const.py` as well as `conf.py`.

FHoevi, 2024-01-07:
@alangibson I would like to contribute by creating a new branch / PR with updated config.py / const.py files which correspond to residential hybrid converters of type SHX.0RT, but I am not allowed to. What do I have to do in that respect?

alangibson, 2024-01-07:
@FHoevi you'll need to fork first then create a branch in your report. You can then create a PR.

Please note that @AlexanderLanin 's repo will probably become the official version of the integration so I'm sure a PR would be appreciated there.

https://github.com/AlexanderLanin/homeassistant-sungrow

FHoevi, 2024-01-09:
@alangibson OK, already downloaded and I will have a look at it first. But it looks a bit different from your repo. Maybe my proposed changes became obsolete? Let's see...

alangibson, 2025-10-08:
Should be fixed in the latest version

closed 2025-10-08 by alangibson