alangibson/homeassistant-sungrow #17: Discussion: fork

AlexanderLanin, 2024-01-06:
I've started a fork to create some pull requests. Which was fine for a while. But much sooner than anticipated my code diverged significantly. I've also addressed this in Discord with @mkraiser and @bohdans, so far there is no clear statement on their side. From the looks of it @bohdans doesn't have much time either way, while @mkaiser might jump on.
Anyway, back to our repos: by now there is not a single line in common between them. That's not intentional! But a byproduct of many many refactorings.

Differences:
* Dropped non modus for now.
* Less model info hardcoded in yaml, more  via auto detection by "groups".
* Expose everything from the model to HA.
* Native support+detection for master/slave.
* Simplify yaml maintanance by removing register info. Also from inverter.py etc.
* Simplify code by shortening functions. (However, overall it's much more code now.)
* Started test suite

Link: https://github.com/AlexanderLanin/homeassistant-sungrow

bohdan-s, 2024-01-06:
Sounds good. Sungather was never going to be a HA integration as it does lots of other things (like PVOutput). 
So happy to support a native integration in any way. 
As for updates I have been working on a rewrite fixing lots of things but it's taking longer than expected.

mkaiser, 2024-01-06:
My YAML-only approach was meant to be a quick integration at first. But it cannot fulfil the requirements most users want to address. A proper HA Integration will also solve most of the current issues in an elegant and scalable way. So I am all-in with my limited ( :/ )time. I can't help much with the coding, but I have gained a quiet good overview of the sugrow systems and their quirks ;)

@AlexanderLanin Do you want to open a discord chan for further discussions?

alangibson, 2024-01-07:
I'm going to start a long term project starting tomorrow so I'm happy letting @AlexanderLanin take the lead. 

I have a couple of notes:
- Let's try to keep one official registers definition yaml. That thing looks like it was a lot of work.
- HTTP is inferior to modbus, but it is still needed for some users. I absolutely cannot get the LAN port to show up on my router to the point I think it's defective. Some forum people have said they can only use the Wifi due to lack of cabling. Also, I've read that the SG routers don't even have the LAN port for modbus.

FHoevi, 2024-05-21:
Dear all, do I get it right that AlexanderLanin's fork is currently the most up-to-date one and furthermore the one which will be maintained in the future?

AlexanderLanin, 2024-05-21:
@FHoevi Those are very different questions. The fork at https://github.com/AlexanderLanin/homeassistant-sungrow is slightly more advanced than the version here. However future maintenance... I'm honestly not sure at the moment.
I'm working on a full rewrite on a branch. Actually, it's like the 5th rewrite... and nothing is working anymore....

For a real productive system, you are better off with either one:
* https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant
* https://github.com/bohdan-s/SunGather

I do believe that the version I have "in progress" has the most features. But I'm not sure if/when I'll invest sufficient time to release it. This is more for contributions than for usage: https://github.com/AlexanderLanin/homeassistant-sungrow/tree/temp