alangibson/homeassistant-sungrow #19: Sungrow SG10RT Configuration Help

freemchr, 2024-02-12:
Hi Guys.

Happy I stumbled upon this. I have a SG10RT and looks like the sensors are collecting information bar a few of them.

I want to set this up in the energy section of home assistant, but I get errors when adding feeds for sensors to:

- Grid Consumption
- Return to Grid
- Solar Production

What are the correct sensors, I should be using? 

If I take a look at the integration the following date points appear to be right.

- sensor.sg10rt_a2313001072_current_active_power shows active solar being generated
- sg10rt_a2313001072_current_export_power shows solar being sent to the grid
- sg10rt_a2313001072_current_load_power shows consumption of my home

Any guidance is appreciated.

freemchr, 2024-02-13:
Have managed to do a bit more digging, and looks like I have found the sensors. But these are not visible in energy.

Grid Consumption - SG10RT A2313001072 Current Load Power
Return to Grid - SG10RT A2313001072 Current Power Exported to Grid
Solar Production - SG10RT A2313001072 Current Active Power

How do I get this into energy?

cartman10, 2024-02-29:
Hi, are there any plans to prvide those cummulative daily totals for the energy dashboard?